nasa/ogma #100: error installation in Dockerfile: `cabal: There is no package named 'alex' `

iftahnaf, 2023-09-21:
Hi,

When trying to install ogma in a Dockerfile, running `cabal v1-install alex happy` results:

```bash
Config file path source is default config file.
Config file /home/ros/.cabal/config not found.
Writing default configuration to /home/ros/.cabal/config
Warning: The package list for '[hackage.haskell.org](http://hackage.haskell.org/)' does not exist. Run 'cabal
update' to download it.
cabal: There is no package named 'alex'.
There is no package named 'happy'.
You may need to run 'cabal update' to get the latest list of available
packages.
```

ivanperez-keera, 2023-09-21:
Thanks! I'll fix this.

By the way, in case it helps, I try to keep this dockerfile up-to-date with every release (I'm selecting the lines that would apply whether you are aiming for fprime, cfs or ros): https://github.com/nasa/ogma/blob/develop-workflows-icarous/workflows/cfs-icarous/Dockerfile#L1-L59

iftahnaf, 2023-09-22:
Thanks.
It helps for the ogma integration in the fprime - ros bridge development environment.

ivanperez-keera, 2023-11-22:
**Description**

The installation instructions in the README do not mention the `cabal update` step. This is confusing users unfamiliar with Haskell, since it's not obvious that cabal should have to be updated before installing Ogma. 

**Type**

- Bug: installation instructions are incomplete.

**Additional context**

None.

**Requester**

- Ivan Perez

**Method to check presence of bug**

The issue is with the documentation. There is no good, simple automated way to check that it is correct.

Because it's not a programming issue, visual inspection is OK.

**Expected result**

The README's installation instructions list the update step prior to installation.

**Desired result**

The README's installation instructions list the update step prior to installation.

**Proposed solution**

Modify the installation instructions in `README.md` to include the cabal update step.

**Further notes**

None.

ivanperez-keera, 2023-11-22:
Technical Lead: Issue scheduled for fixing in Ogma 1.0.12.

Fix assigned to: @ivanperez-keera .

ivanperez-keera, 2023-11-22:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2023-11-22:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles passes the tests. Details:
    Docker image:
    ```dockerfile
    FROM ubuntu:trusty
    
    RUN apt-get update
    
    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update
    
    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev
    
    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH
    
    RUN cabal update
    RUN cabal v1-sandbox init
    RUN cabal v1-install alex happy
    RUN apt-get install --yes git
    
    # We install the application first and then test it, in case any tests need to
    # run the tool.
    CMD git clone $REPO && \
        cd $NAME && \
        git checkout $COMMIT && \
        cd .. && \
        cabal v1-install $NAME/$PAT**/ --enable-tests && \
        cabal v1-install $NAME/$PAT**/ --enable-tests --run-tests -j1
    ```
    Command:
    ```sh
    $ docker run -e "REPO=https://github.com/ivanperez-keera/ogma" -e "NAME=ogma" -e PAT="ogma-" -e "COMMIT=94a59586aaa46765fccc67d327485940154fa811" -it ogma-test
    ```

  - [X] The solution proposed fixes the issues described. Details:
    The issue is with the documentation. We verify the solution proposed by visual inspection.
- [X] Implementation is documented. Details:
  No changes needed.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No examples affected.
- [X] Required version bumps are evaluated. Details:
  Bump not needed (change to documentation only).

ivanperez-keera, 2023-11-22:
Change Manager: Implementation ready to be merged.

commit a56a2f8d (2023-11-22): Merge branch 'develop-readme-cabalUpdate' into develop. Close #100.

**Description**

The installation instructions in the README do not mention the `cabal update`
step. This is confusing users unfamiliar with Haskell, since it's not obvious
that cabal should have to be updated before installing Ogma.

**Type**

- Bug: installation instructions are incomplete.

**Additional context**

None.

**Requester**

- Ivan Perez

**Method to check presence of bug**

The issue is with the documentation. There is no good, simple automated way to
check that it is correct.

Because it's not a programming is [... cut]
commit d4292671 (2023-11-22): Merge branch 'develop-readme-cabalUpdate' into develop. Close #100.

**Description**

The installation instructions in the README do not mention the `cabal update`
step. This is confusing users unfamiliar with Haskell, since it's not obvious
that cabal should have to be updated before installing Ogma.

**Type**

- Bug: installation instructions are incomplete.

**Additional context**

None.

**Requester**

- Iftach Naftaly (Lulav Space)

**Method to check presence of bug**

The issue is with the documentation. There is no good, simple automated way to
check that it is correct.

Because it's no [... cut]
closed 2023-11-22 by ivanperez-keera