nasa/ogma #129: include error when using ogma ros with fret files

EliasH-NMBU, 2024-03-05:
When running
cabal v1-exec -- runhaskell demo.hs
I receive a fret.c and fret.h file (and a fret_types.h)
The minor thing is that the ogma ros files expect these to be named monitor.h and monitor.c.
Meaning each time I generate a ogma ros package I have to change the include names. 
Just wondered if it could be changed so that the .hs file says "monitor" instead of "fret" 

FROM
main :: IO ()
main = reify spec >>= compile "fret"

TO
main :: IO ()
main = reify spec >>= compile "monitor"

ivanperez-keera, 2024-03-05:
Hi Elias!

There is a way to do this. The argument `--target-file-name` allows you to control this setting. Use `--help` if unsure.

ivanperez-keera, 2024-03-05:
I'm closing as not needed because the feature is there, but feel free to re-open if that doesn't work.

closed 2024-03-05 by ivanperez-keera