nasa/ogma #213: `ogma-extra`: Upgrade vesion bounds of dependency on `Cabal`

ivanperez-keera, 2025-01-23:
ogma-extra wants Cabal < 3.9 whereas Debian has Cabal 3.10.3.0.

ivanperez-keera, 2025-01-24:
Technical Lead: Confirmed that the issue should be addressed.

kaBeech, 2025-02-01:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2025-02-01:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    - Build logs:
      - https://github.com/nasa/ogma/actions/runs/13083404478/job/36512965346?pr=223
      - https://github.com/nasa/ogma/actions/runs/13083404473/job/36512965347?pr=223
    - Build log archives:
    - [logs_33797562808.zip](https://github.com/user-attachments/files/18627979/logs_33797562808.zip)
    - [logs_33797562815.zip](https://github.com/user-attachments/files/18627980/logs_33797562815.zip)
  - [X] The solution proposed produces the expected result. Details:
    The following dockerfile installs Ogma using `Cabal >= 3.14`, after which it prints the message "Success":
    ```Dockerfile
    FROM ubuntu:focal
    
    RUN apt-get update
    
    RUN apt-get install --yes \
          curl g++ gcc git libgmp3-dev libz-dev make
    
    RUN mkdir -p $HOME/.local/bin
    ENV PATH=$PATH:/root/.local/bin/
    
    RUN curl https://downloads.haskell.org/~ghcup/0.1.17.7/x86_64-linux-ghcup-0.1.17.7 -o $HOME/.local/bin/ghcup
    RUN chmod a+x $HOME/.local/bin/ghcup
    
    ENV PATH=$PATH:/root/.ghcup/bin/
    RUN ghcup install ghc 9.6
    RUN ghcup install cabal 3.12
    RUN ghcup set ghc 9.6.6
    RUN cabal update
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO \
        && cd $NAME \
        && git checkout $COMMIT \
        && cabal v1-install alex happy \
        && cabal v1-install BNFC \
        && cabal v1-install ogma-**/ --constraint='Cabal>=3.14' \
        && echo Success
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/kaBeech/ogma" -e "NAME=ogma" -e "COMMIT=26276b51471ab57fbc004e37fa0c946479b6323a" -it ogma-verify-213
    ```
- [X] Implementation is documented. Details:
  No updates needed; change is to version numbers only.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed; change is to version numbers only.
- [X] Author is internal or has provided signed CLA.
- [X] Required version bumps are evaluated. Details:
   Bump not needed; change extends, but does not restrict, the range of versions compatible with Copilot.

ivanperez-keera, 2025-02-01:
Change Manager: Implementation ready to be merged.