nasa/ogma #222: `ogma-cli`: template variables in README are formatted incorrectly

ivanperez-keera, 2025-01-29:
**Description**

The formatting of template variables in the ["Template Customization"](https://github.com/nasa/ogma?tab=readme-ov-file#template-customization-2) section of the F' backend is incorrect. They are being rendered as plain text, when they should use backticks to be formatted as code.

**Type**

- Bug: Incorrect formatting in documentation.

**Additional context**

None.

**Requester**

- Ivan Perez.

**Method to check presence of bug**

The following regexp search lists all occurrences of template variables and their immediately surrounding characters, filtering out the ones that have backticks:
```
$ grep -nHe '.{\+[a-zA-Z]\+}\+.' ogma-cli/README.md | grep -ve '`{\+.*}\+`'
ogma-cli/README.md:491:- {{{ifaceTypePorts}}}: Contain the type declarations for the types used by the
ogma-cli/README.md:494:- {{{ifaceInputPorts}}}: Contains the declarations of the `async input` port,
ogma-cli/README.md:497:- {{{ifaceViolationEvents}}}: Contains the output port declarations, used to
ogma-cli/README.md:502:- {{{hdrHandlers}}}: Contains the declarations of operations to execute when
ogma-cli/README.md:508:- {{{implInputs}}}: Contains the declarations of the variables with inputs
ogma-cli/README.md:511:- {{{implMonitorResults}}}: Contains the declarations of boolean variables,
ogma-cli/README.md:514:- {{{implInputHandlers}}}: Contains the implementations of operations to
ogma-cli/README.md:518:- {{{implTriggerResultReset}}}: Contains instructions that reset the status of
ogma-cli/README.md:521:- {{{implTriggerChecks}}}: Contains instructions that check whether any monitor
ogma-cli/README.md:525:- {{{implTriggers}}}: Contains the implementations of the functions that set
```

**Expected result**

The search above is empty, indicating that all template variables in the documentation are surrounded by backticks.

**Desired result**

The search above is empty, indicating that all template variables in the documentation are surrounded by backticks.

**Proposed solution**

Modify README to surround all template variables with backticks.

**Further notes**

None.

ivanperez-keera, 2025-02-04:
Technical Lead: Issue scheduled for fixing in Ogma 1.7.0.

Fix assigned to: @carte731.

carte731, 2025-02-04:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2025-02-05:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    - Build logs:
      - https://github.com/nasa/ogma/actions/runs/13146415071/job/36685448189
      - https://github.com/nasa/ogma/actions/runs/13146415108/job/36685448186
    - Build log archives:
      - [logs_33943669676.zip](https://github.com/user-attachments/files/18664725/logs_33943669676.zip)
      - [logs_33943669762.zip](https://github.com/user-attachments/files/18664726/logs_33943669762.zip)
  - [X] The solution proposed produces the expected result. Details:
    The following dockerfile checks that there are no template variables in the README that are not properly formatted, after which it prints the message "Success":
    ```Dockerfile
    FROM ubuntu:focal
    
    RUN apt-get update
    
    RUN apt-get install --yes git
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO \
        && cd $NAME \
        && git checkout $COMMIT \
        && ! (grep -nHe '.{\+[a-zA-Z]\+}\+.' ogma-cli/README.md | grep -ve '`{\+.*}\+`') \
        && echo "Success"
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e "REPO=https://github.com/carte731/ogma" -e "NAME=ogma" -e "COMMIT=0cddcd0653cbcf65eb18466b2b28e322a3b331d1" -it ogma-verify-222
    ```
- [X] Implementation is documented. Details:
  No updates needed; change is to the documentation.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed; change is to the documentation.
- [X] Author is internal or has provided signed CLA.
- [X] Required version bumps are evaluated. Details:
  Bump not needed; change is to the documentation.

ivanperez-keera, 2025-02-05:
Change Manager: Implementation ready to be merged.

commit 0a4a9ce4 (2025-02-05): Merge branch 'develop-fix-readme' into develop. Close #222.

**Description**

The formatting of template variables in the ["Template
Customization"](https://github.com/nasa/ogma?tab=readme-ov-file#template-customization-2)
section of the F' backend is incorrect. They are being rendered as plain text,
when they should use backticks to be formatted as code.

**Type**

- Bug: Incorrect formatting in documentation.

**Additional context**

None.

**Requester**

- Ivan Perez.

**Method to check presence of bug**

The following regexp search lists all occurrences of template variables and
their imme [... cut]
closed 2025-02-05 by ivanperez-keera