nasa/ogma #258: `ogma-core` Tests are missing auxiliary files

swt2c, 2025-02-21:
`ogma-core` 1.6.0 faces a similar problem as in #216.

```
Running 1 test suites...
Test suite unit-tests: RUNNING...
standalone-cmd-fcs-ok: [Failed]
The result of the transformation of input file tests/fcs_good.json to Copilot was unexpected.
standalone-cmd-fsc-file-not-found: [OK]
standalone-cmd-fcs-parse-fail-1: [OK]
standalone-cmd-fcs-parse-fail-2: [OK]
standalone-cmd-fcs-parse-fail-3: [OK]
standalone-reqs-db-cocospec: [Failed]
The result of the transformation of input file tests/fdb-example1.json to Copilot was unexpected.
structs-parse-ok: [Failed]
The result of the transformation of the C header file tests/reduced_geofence_msgs.h to Copilot struct declarations was unexpected.
structs-parse-fail-1: [OK]

         Test Cases  Total 
 Passed  5           5     
 Failed  3           3     
 Total   8           8     
Test suite unit-tests: FAIL
Test suite logged to: dist-ghc/test/ogma-core-1.6.0-unit-tests.log
0 of 1 test suites (0 of 1 test cases) passed.
```

ivanperez-keera, 2025-03-22:
**Description**

The cabal file for `ogma-core` is not including all the files needed for the tests, causing the tests to fail in some environments.

**Type**

- Bug: Files missing in package.

**Additional context**

None.

**Requester**

- Ivan Perez.

**Method to check presence of bug**

In some environment, running the tests from `ogma-core` reports errors due to missing files:

```
[...]
Running 1 test suites...
Test suite unit-tests: RUNNING...
standalone-cmd-fcs-ok: [Failed]
The result of the transformation of input file tests/fcs_good.json to Copilot was unexpected.
standalone-cmd-fsc-file-not-found: [OK]
standalone-cmd-fcs-parse-fail-1: [OK]
standalone-cmd-fcs-parse-fail-2: [OK]
standalone-cmd-fcs-parse-fail-3: [OK]
standalone-reqs-db-cocospec: [Failed]
The result of the transformation of input file tests/fdb-example1.json to Copilot was unexpected.
structs-parse-ok: [Failed]
The result of the transformation of the C header file tests/reduced_geofence_msgs.h to Copilot struct declarations was unexpected.
structs-parse-fail-1: [OK]

         Test Cases  Total 
 Passed  5           5     
 Failed  3           3     
 Total   8           8     
Test suite unit-tests: FAIL
Test suite logged to: dist-ghc/test/ogma-core-1.6.0-unit-tests.log
0 of 1 test suites (0 of 1 test cases) passed.
```

Unfortunately, this error is not easy to replicate, due to some files being automatically included by cabal in some cases even when they are not mentioned as `extra-source-files`.

**Expected result**

Installation from the package archives completes successfully.

**Desired result**

Installation from the package archives completes successfully.

**Proposed solution**

Add any auxiliary files needed for tests to the cabal package declaration of `ogma-core`.

**Further notes**

None.

ivanperez-keera, 2025-03-22:
Change Manager: Confirmed that the issue exists.

Confirmation carried out by external tester.

ivanperez-keera, 2025-03-22:
Technical Lead: Issue scheduled for fixing in Ogma 1.7.0.

Fix assigned to: @ivanperez-keera .

ivanperez-keera, 2025-03-22:
Implementor: Solution implemented, review requested.

ivanperez-keera, 2025-03-22:
Change Manager: Verified that:
- Solution is implemented:
  - [X] The code proposed compiles and passes all tests. Details:
    - Build logs:
      - https://github.com/nasa/ogma/actions/runs/14007344460/job/39222897660
      - https://github.com/nasa/ogma/actions/runs/14007344459/job/39222897655
    - Build log archives:
      - [logs_36063417930.zip](https://github.com/user-attachments/files/19403995/logs_36063417930.zip)
      - [logs_36063417935.zip](https://github.com/user-attachments/files/19403996/logs_36063417935.zip)
  - [X] The solution proposed produces the expected result. Details:
    The following Dockerfile installs `ogma-core`, creates a distributable source package, removes the `ogma-core` directory, unpacks the version from the distributable package, and runs the test from that unpacked archive, after which it prints the message "Success":
    ```Dockerfile
    FROM ubuntu:focal
    
    ENV DEBIAN_FRONTEND=noninteractive
    
    RUN apt-get update
    
    RUN apt-get install --yes software-properties-common
    RUN add-apt-repository ppa:hvr/ghc
    RUN apt-get update
    
    RUN apt-get install --yes ghc-8.6.5 cabal-install-2.4
    RUN apt-get install --yes libz-dev alex happy
    RUN apt-get install --yes git g++ libbz2-dev libexpat-dev
    
    ENV PATH=/opt/ghc/8.6.5/bin:/opt/cabal/2.4/bin:$PWD/.cabal-sandbox/bin:$PATH
    
    RUN cabal update
    RUN mkdir -p /root/test
    WORKDIR /root/test/
    
    SHELL ["/bin/bash", "-c"]
    CMD git clone $REPO \
        && cd $NAME \
        && cabal v1-sandbox init --sandbox=$PWD/../.cabal-sandbox \
        && git checkout $COMMIT \
        && cabal v1-install BNFC --constraint='happy<2.0' \
        && cabal v1-install ogma**/ --constraint='aeson>=2.0.3.0' --enable-tests \
        && cd ogma-core \
        && cabal sdist \
        && cp dist/ogma-core*.tar.gz .. \
        && cd .. \
        && rm -rf ogma-core \
        && tar -zxvpf ogma-core*.tar.gz \
        && mv ogma-core-1.6.0 ogma-core \
        && cd ogma-core \
        && find \
        && cabal v1-sandbox --sandbox=$PWD/../../.cabal-sandbox/ init \
        && cabal v1-install --constraint='aeson>=2.0.3.0' --enable-tests \
        && cabal v1-test \
        && echo Success
    ```
    Command (substitute variables based on new path after merge):
    ```
    $ docker run -e REPO=https://github.com/ivanperez-keera/ogma -e COMMIT=a033a3e5e7163ab4579ee17c90663c0df01725f9 -e NAME=ogma ogma-verify-258
    ```
- [X] Implementation is documented. Details:
  No updates needed; no new top-level definitions or introduction of new code.
- [X] Change history is clear.
- [X] Commit messages are clear.
- [X] Changelogs are updated.
- [X] Examples are updated. Details:
  No updates needed; changes are to the cabal files.
- [X] Author is internal or has provided signed CLA.
- [X] Required version bumps are evaluated. Details:
  Bump not needed; change does not affect API or user interface.

ivanperez-keera, 2025-03-22:
Change Manager: Implementation ready to be merged.

commit df588b3f (2025-03-22): Merge branch 'develop-core-tests' into develop. Close #258.

**Description**

The cabal file for `ogma-core` is not including all the files needed for
the tests, causing the tests to fail when the package is installed from
hackage.

**Type**

- Bug: Files missing in package.

**Additional context**

None.

**Requester**

- Scott Talbert.

**Method to check presence of bug**

When the package is installed from hackage, running the tests from `ogma-core` reports errors due to missing files:

```
[...]
Running 1 test suites...
Test suite unit-tests: RUNNING...
standalone-cmd-fcs-ok: [Failed]
The  [... cut]
closed 2025-03-22 by ivanperez-keera